cmaglie/FlashStorage #1: What's the point of using this library vs. simply defining variables?

q2dg, 2015-10-09:
I mean...if Flash memory will be erased when a new sketch is uploaded...I can't see the usefulness of this library. Thanks.

q2dg, 2015-10-09:
Ooh, I see...PROGMEM is only valid in AVR, so this library would be like a "replacement" of this feature, isn t?
Sorry for bothering you

closed 2015-10-09 by q2dg

cmaglie, 2015-10-10:
> Ooh, I see...PROGMEM is only valid in AVR, so this library would be like a "replacement" of this feature, isn t?

Well not exactly.

The equivalent of PROGMEM for ARM is to just declare data as `static const`, since ARM Cortex-M has a flat addressing space and you can access flash memory directly, you don't need to pass through all the `pgmspace.h` mess like on AVR.

```C++
const char flash_string[] = "Hello!";   // <-- On 32-bit ARM Cortex this is stored in Flash

void loop() {
     Serial.println(flash_string);
}
```

This is far easier compared to AVRs!

On the other hand most Cortex-M based CPU doesn't provide an **EEPROM** (well, some chip-maker may have some CPU models with it, but from my experience it's not so common) so this library is aimed to be an alternative to EEPROM (in fact allows you to **change** the content of the data stored in flash).

Of course this library shows also the limits of a Flash memory compared to an EEPROM:

* Flash memory must be written in "pages" and those pages must be erased before write: if you need to change just 1 byte you must erase + write a block of 256 bytes (a page) while with an EEPROM you can just write 1 byte.
* Flash memory wears out quickly compared to an EEPROM: a typical flash memory can perform between 20k..50k erase cycles before starting to lose ability to retain data, an EEPROM can do 10x that.

q2dg, 2015-10-11:
Oooh, I see! Thanks a lot for your explanation! Maybe it should be placed in README.md, as it clears several doubts other people could also have. Thanks!