cmaglie/FlashStorage #15: Wrong default values?

tuxedo0801, 2017-08-15:
A fresh and unused eeprom has usually 0xff on each byte... The FlashAsEEPROM gives you 0x00....

Maybe this should be changed? What do you think?

cmaglie, 2017-09-04:
Hi @tuxedo0801 

how can we do that?
The only way that comes to mind is to write 0xFF on the whole EEPROM area, once at the first run of the sketch, but we need to reserve a bit to save also this state on a off-EEPROM area.

tuxedo0801, 2017-09-04:
Hmm, your're right, there needs to be an extra bit for storing this state...
But maybe we then just provide this extra bit via API, so that the user can decide what to do?!

cmaglie, 2017-09-04:
I see that there is already a `valid` flag:

https://github.com/cmaglie/FlashStorage/blob/master/src/FlashAsEEPROM.h#L33

it could be used here:

https://github.com/cmaglie/FlashStorage/blob/master/src/FlashAsEEPROM.cpp#L53

so, after the `_eeprom = eeprom_storage.read();` we can check if `_eeprom.valid` is `false`, in that case we should fill the `_eeprom.data` with `0xff` no need to write the flash, something like:

```C++
void EEPROMClass::init()
{
  _eeprom = eeprom_storage.read();
  if (!_eeprom.valid) {
    memset(_eeprom.data, 0xFF, EEPROM_EMULATION_SIZE);
  }
  _initialized = true;
}
```

I don't have the time to try this solution right now, if you want to give it a go you're welcome (and maybe open also a PR... :-)).

tuxedo0801, 2017-09-04:
Sounds like a good idea. I will test it and create PR

tuxedo0801, 2017-09-04:
see #16

cmaglie, 2017-10-06:
Fixed by #16

closed 2017-10-06 by cmaglie

tuxedo0801, 2017-10-06:
Thanks a lot.