cmaglie/FlashStorage #19: flash endurance > 10k

mattvenn, 2018-03-28:
Hi, 
first off, thanks for the library. This is more of a comment than an issue, but perhaps it would be good to update the README.

You're giving the number 10k, which I was slightly worried about in my application - so I checked the datasheet: 36.10 NVM characteristics. Flash as EEPROM emulation is minimum 100k typical 600k cycles. 

I'm now running a burn test on a SAMD21J, and it is at 500k cycles so far with no errors (which is why I checked the datasheet). I'll edit this comment to add when I start seeing errors.

mattvenn, 2018-04-23:
Quick update - my burn test has run for 2M writes across 80 addresses, still no errors.

I also added a pin toggle to the write and erase methods to check that they were really happening.

Yellow trace is erase and blue is write.
[image]

cmaglie, 2018-04-23:
Hi @mattvenn,
the "scary" message in the readme has been made with the specific purpose to let the user focus on an aspect that is easily overlooked: the flash memory can wear out.

I wrote the number 10.000, instead of 100.000 or more, because in a real application it can be easily underestimated the **total** number of writes, especially if the life of the microcontroller spans over many years.

In your case you did the right thing: you checked the datasheet and performed a test on a real device. Maybe you will have done it anyways, maybe not, but surely the "scary" message pushed you do this test and stay on the safe side.

Now I'm courious to see what's the "magic number" :-)

mattvenn, 2018-04-23:
I'll keep going until it breaks and let you know!

closed 2018-08-09 by cmaglie