cmaglie/FlashStorage #20: Error handling or return value?

thkaw, 2018-04-30:
Hi, this is nice libary.

But I found if I build a simple struct like:
```
struct sct_script {
	char content[40][650];
} sct_actionScript;

FlashStorage(script_store, sct_script);

```

but if read the script_store before the script_store not write anything yet.

It cause arduino freeze and not able keep running until reset manually.

I tracking the code find read using memcpy(data, (const void *)flash_ptr, size);

but I have no idea to handling memcpy return status code.

cmaglie, 2018-08-09:
Whould you like to share a complete sketch?

BTW it looks like that the `content` array is very big `40*650 = 26000` it tooks more than 26KB of RAM, are you sure you're not running out of memory?

thkaw, 2018-09-17:
Sorry for late replied,
I'm using ATSAMD21 which has 32KB RAM, 

I notice the freeze behavior may cause by out of memory too.

So that I just attach EEPROM IC to solve the memory size problem.

closed 2018-09-17 by thkaw