cmaglie/FlashStorage #21: Any chance to port to SAMD51?

BrunDog, 2018-05-21:


cmaglie, 2018-08-09:
Hi @BrunDog 
Unfortunately I don't have a board with a SAMD51 at hand...
Anyway, since it a device of the same SAMD family maybe there is a chance that the library works as is? Did you have tried already?

BrunDog, 2018-08-09:
Actually, it won't even compile. It throws multiple exceptions. I am using the Adafruit Metro M4. You could try compiling for this board to see the errors.

BrunDog, 2018-09-07:
Touching base again to see if you could look at this.

BrunDog, 2018-09-17:
Hi Cristian,

 

Sorry to be pesky… I am trying to use a board with this micro, but no joy since Flash storage is needed in my implementation. Any chance you could take a look – if you can try to figure out the compile-time errors (I think it is in the board definitions), I am happy to run the tests.

 

Best regards,

 

Peter Bruno

 <mailto:PBruno3@gmail.com> PBruno3@gmail.com

M: 954-899-8100

PaulZC, 2018-09-17:
Hi there,

I can duplicate the problems @BrunDog is having. I've tried compiling the FlashStoreAndRetrieve example. Works fine when I select the Adafruit Feather M0 (ATSAMD51G18), but when I select the Adafruit Metro M4 (ATSAMD51J19) I get the errors included below. Looks like some of the NVMCTRL definitions are missing for the M4? Too deep for me. A subtle problem in CMSIS nvmctrl.h perhaps?

I've updated both my Arduino SAMD boards and my Adafruit SAMD boards - made no difference...

Cheers,

Paul

Arduino: 1.8.5 (Windows 10), Board: "Adafruit Metro M4 (SAMD51), Enabled"

\\thunderbird\home\pclark\Documents\Arduino\libraries\FlashStorage-master\src\FlashStorage.cpp: In member function 'void FlashClass::write(const volatile void*, const void*, uint32_t)':

\\thunderbird\home\pclark\Documents\Arduino\libraries\FlashStorage-master\src\FlashStorage.cpp:56:22: error: 'volatile struct NVMCTRL_CTRLB_Type::<anonymous>' has no member named 'MANW'

   NVMCTRL->CTRLB.bit.MANW = 1;

                      ^

\\thunderbird\home\pclark\Documents\Arduino\libraries\FlashStorage-master\src\FlashStorage.cpp:61:26: error: 'NVMCTRL_CTRLA_CMDEX_KEY' was not declared in this scope

     NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_PBC;

                          ^

\\thunderbird\home\pclark\Documents\Arduino\libraries\FlashStorage-master\src\FlashStorage.cpp:61:52: error: 'NVMCTRL_CTRLA_CMD_PBC' was not declared in this scope

     NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_PBC;

                                                    ^

\\thunderbird\home\pclark\Documents\Arduino\libraries\FlashStorage-master\src\FlashStorage.cpp:62:33: error: 'volatile struct NVMCTRL_INTFLAG_Type::<anonymous>' has no member named 'READY'

     while (NVMCTRL->INTFLAG.bit.READY == 0) { }

                                 ^

\\thunderbird\home\pclark\Documents\Arduino\libraries\FlashStorage-master\src\FlashStorage.cpp:74:52: error: 'NVMCTRL_CTRLA_CMD_WP' was not declared in this scope

     NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_WP;

                                                    ^

\\thunderbird\home\pclark\Documents\Arduino\libraries\FlashStorage-master\src\FlashStorage.cpp:75:33: error: 'volatile struct NVMCTRL_INTFLAG_Type::<anonymous>' has no member named 'READY'

     while (NVMCTRL->INTFLAG.bit.READY == 0) { }

                                 ^

\\thunderbird\home\pclark\Documents\Arduino\libraries\FlashStorage-master\src\FlashStorage.cpp: In member function 'void FlashClass::erase(const volatile void*)':

\\thunderbird\home\pclark\Documents\Arduino\libraries\FlashStorage-master\src\FlashStorage.cpp:93:24: error: 'NVMCTRL_CTRLA_CMDEX_KEY' was not declared in this scope

   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_ER;

                        ^

\\thunderbird\home\pclark\Documents\Arduino\libraries\FlashStorage-master\src\FlashStorage.cpp:93:50: error: 'NVMCTRL_CTRLA_CMD_ER' was not declared in this scope

   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_ER;

                                                  ^

\\thunderbird\home\pclark\Documents\Arduino\libraries\FlashStorage-master\src\FlashStorage.cpp:94:32: error: 'volatile struct NVMCTRL_INTFLAG_Type::<anonymous>' has no member named 'READY'

   while (!NVMCTRL->INTFLAG.bit.READY) { }

                                ^

exit status 1
Error compiling for board Adafruit Metro M4 (SAMD51).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

BrunDog, 2018-11-06:
Pinging once again!

mitchellpontague, 2018-12-11:
#27 I've made a pull request for this issue, from the datasheet the registers I've updated look correct but could use testing.

BrunDog, 2018-12-21:
Worked!! Brilliant!!

THANK YOU! @cmaglie can pull it!

mitchellpontague, 2018-12-21:
I've not had the same success yet, but it could be unrelated.

BrunDog, 2018-12-21:
I edited my comment above.

BrunDog, 2019-01-15:
Hi... any help so appreciated!

rossfl, 2019-01-18:
> Hi @BrunDog
> Unfortunately I don't have a board with a SAMD51 at hand...
> Anyway, since it's a device of the same SAMD family maybe there is a chance that the library works as is? Did you have tried already?

I too could really use your utility on the SAMD51.. I would happy to send you a metroM4 express board if that would help?

trentben, 2019-01-23:
Same issue here, using a feather wing M4. Support for SAMD51 would be much appreciated

mitchellpontague, 2019-01-25:
I'm looking into this again, i believe the issue is the larger erase granularity hammering code addresses. I've updated my pull request if someone wants to try again. For me it is working with a simple integer increment.

trentben, 2019-02-02:
@mitchellpontague Gave your fork a try, but my board crashes just by including FlashStorage.h
I'm using a Featherwing M4 Express.

smarpug, 2019-02-19:
I am using an ItsyBitsy M4 Express (ATSAMD51), and this library is working so far for me (without the cache enabled; I haven't tried it with cache, but I'm not confident it would work). However, the program size being reported at compile time is 60% of the (512K) flash memory. Same program compiled on the ItsyBitsy M0 Express (ATSAMD21) with 13% of 256K less flash memory.
I'm looking into it, but at first glance my initial guess is that it may have to do with where in the physical memory the FlashStorage items are being allocated in relation to the program code.

samuelBrown6, 2019-02-21:
@smarpug Could you share how you pulled this off?  I'm using the same chip and am getting errors.

seawarrior181, 2019-07-30:
@smarpug, @samuelBrown6, I, too am using the ItsyBitsy M4 Express and I can't get this library to work (it compiles with the same errors cited by @PaulZC, above.  Any advice?

smarpug, 2019-07-30:
Sorry to miss your comment samuelBrown6; I was actually using the code from pull request #27, which I should have noted; I got the threads mixed up a little here. Sorry about that. Additionally, I was not using EEPROM emulation, so if you are using that side of the code, I can't speak to that. I've actually got too many sticks in the fire and haven't actually taken the time to dig much further into this yet, but at last try the code from #27 did compile for me, albeit in a program size that was far larger than it should have been.

seawarrior181, 2019-07-31:
@smarpug, thanks for the reply.  I was actually able to get it to compile with your code from #27 (after posting my question), and, strangely, it didn't adversely affect the program size as reported by @samuelBrown6.

samuelBrown6, 2019-07-31:
I also got it working using #27 , thanks for the help!

smarpug, 2019-08-05:
I think the issue I was having with the large program size may have to do with the macro: 
`#define FlashStorage(name, T) \
  __attribute__((__aligned__(8192))) \
  static const uint8_t PPCAT(_data,name)[(sizeof(T)+8191)/8192*8192] = { }; \
  FlashStorageClass<T> name(PPCAT(_data,name));`
I had a large number of individual integer flash storage objects (apparently a very bad way of doing this), and each one would appear to take a significant number of bytes (~8196 bytes for a 32-bit int).
Easy fix for me will be to put most of my integers in a single struct before passing that struct to the FlashStorage object.
My poor implementation aside, this pull request has tested well for me.
Edit: I can now confirm that consolidating my objects into container structs reduced the program size considerably.

cmaglie, 2020-04-02:
fixed by #27

closed 2020-04-02 by cmaglie