cmaglie/FlashStorage #31: EEPROM storage allocated even though it is not used

sslupsky, 2019-10-11:
I noticed the following two declarations in EEPROM.cpp

https://github.com/cmaglie/FlashStorage/blob/18c021255ca8b6962bbcdd9b1f60a6ed6f2311e0/src/FlashAsEEPROM.cpp#L24

https://github.com/cmaglie/FlashStorage/blob/18c021255ca8b6962bbcdd9b1f60a6ed6f2311e0/src/FlashAsEEPROM.cpp#L74

This allocates quite a bit of RAM (1K) when you are not using the EEPROM emulation.

Would it be advantageous to have an initialization method (ie: EEPROM:begin() ) that is called to allocate the storage?

sslupsky, 2019-10-11:
There is already an EEPROM::init() method.  Perhaps the allocation could be done there?

The way the FlashStorage define works and the FlashStorageClass is templated, it is not obvious how to move the allocation into a class.  I was trying do so using my own class and encountered compiler error.  The declaration of the _data array is declared as static storage class and this causes problems with the FlashStorageClass template.

Is there a way to instantiate the FlashStorageClass object from within a class?  I have two different types of storage that I would like to create two different FlashStorageClass objects for.

sslupsky, 2019-10-12:
I defined EEPROM_EMULATION_SIZE 0 as follows:

```
#define BOOT_STATUS_FILE_MAX_RECORDS 16
#include <FlashStorage.h>

```
but for some reason that does not appear to prevent the allocation of the memory.  Here are the relevant allocations from nm:

```
20000d6c 00000403 B EEPROM
00017800 00000500 t _dataeeprom_storage

```
There is 1K of RAM and 1.2K of FLASH allocated to the EEPROM storage.

Note that I do not include FlashAsEEPROM.h.  So, it appears to me that FlashAsEEPROM.cpp is being compiled and linked regardless of its inclusion.  Does this imply that FlashAsEEPROM.* need to be in their own separate library or is there a way to prevent them from being compiled when not used?

tuxedo0801, 2020-09-27:
interesting question...