cmaglie/FlashStorage #35: Any chance to port to nRF52840?

rmlearney-digicatapult, 2020-03-25:
@cmaglie would it be possible to use this as a basis for a library to read/write internal flash on the new Nano 33 BLE boards?

I've got a project where I'd like to set a non-volatile 'dirty' bit in flash to check for clean shutdowns.

marcel-bluestone, 2020-05-25:
I'd also be interested in using the lib with Nano 33 BLE boards....

pwielgosz, 2020-07-31:
The BLE boards really do need some sort of on-board ability to save parameters.   With all the built in items such as mag meters, accelerometers, bluetooth, etc there are many one chip projects that  can be created.  Unfortunately without an EEPROM or emulation we are forced to add additional hardware to do simple things like save project configuration/calibration data.  Really need this functionallity.
I should also point out that the EEPROM emulation library for FlashStorage says this supports the Arduino 33 BLE and in fact it does not.  Another case of it should work but actually does not.
Sadly these boards always seem to have a roadblock when doing projects because something basic or common is not supported (EEPROM, one-wire).  Knowing what I know now I would run screaming from these boards knowing its much safer to piece together the hardware manually but get full functionality and control.

rmlearney-digicatapult, 2020-08-03:
I agree. Observing how Arduino has worked over the years, it always looks like they are too quick to move to releasing the next device before the features on existing ones are complete.

Right now, the team is focusing on the Portenta at the expense of completing the features of the (less than 1 year old) Arduino 33 BLE - this include BLE library stability, improving documentation, adding code samples for functionality, adding an EEPROM library, etc.

This approach to the product cycle means nobody can be assured of functionality when buying even the latest top end device.

pwielgosz, 2020-08-03:
My single board With a few LEDs project just expanded with a silly 60 cent chip (I2C EEPROM) so that I could simply save some calibration parameters.  Will try to make the 8-pin chip like a shield so that I have an EEPROM available on future projects.  The Accel is great, I use the mag sensor for button inputs (can sense multiple directions and strokes) just needs the basics added the board.