cmaglie/FlashStorage #37: SAMD51 NVM cache disable before write

dhalbert, 2020-04-02:
@mitchellpontague @cmaglie 

I didn't have a chance to look at #27 before it was merged. Here is one comment.

This erratum says you need to disable the NVM cache before writing to the Page Buffer, for Rev A EFP and Rev D chips. I had misinterpreted the erratum as applicable to old Rev A chips only and had commented this out in our code. In practice I haven't seen this issue, but better safe than sorry.

[image]

cmaglie, 2020-04-02:
Thanks @dhalbert this explains why #27 wasn't working when cache is enabled.
Do you have a patch for that already?

dhalbert, 2020-04-02:
There are two caches: the CMCC cache and the NVMCTRL cache. The NVMCTRL cache is enabled on reset; the CMCC cache is disabled on reset.

I'll do some testing to confirm which affects this. We are turning off the NVMCTRL cache in the UF2 bootloader before writing. We don't enable the CMCC cache in the UF2 bootloader, but we do in CircuitPython.

PR #38 (2020-05-04): handle SAMD51 caches
Fixes #37 and cache issues noted in #27.
- Temporarily disable NVMCTRL line caches when writing, per SAMD51 errata. Restore them to original settings when done.
- Invalidate CMCC cache after writing if it was in use .

Tested on an early SAMD51, which might have been more likely to have the cache issues. Without the CMCC invalidation, the test program in #27 returned incorrect results, as noted by some commenters in that issue.
closed 2020-05-04 by dhalbert