cmaglie/FlashStorage #46: typo , should be SR but wrote CR

WinflyLam, 2020-10-10:
typo on line 56 of FlashStorage.cpp
```
#if defined(__SAMD51__)
// Invalidate all CMCC cache entries if CMCC cache is enabled.
static void invalidate_CMCC_cache()
{
  if (CMCC->CR.bit.CSTS) {      <--- SR.bit.CSTS
    CMCC->CTRL.bit.CEN = 0;
    while (CMCC->SR.bit.CSTS) {}
    CMCC->MAINT0.bit.INVALL = 1;
    CMCC->CTRL.bit.CEN = 1;
  }
}
#endif
```

obra, 2021-01-02:
@dhalbert Do you have opinions on this? I'm unable to build with #38  merged:

```
/Users/jesse/Documents/Arduino/libraries/FlashStorage/src/FlashStorage.cpp:56:13: error: 'struct Cmcc' has no member named 'CR'; did you mean 'SR'?
   56 |   if (CMCC->CR.bit.CSTS) {
      |             ^~
      |             SR
```

dhalbert, 2021-01-02:
This fix is correct. I looked at my old work on this and it appears that I fixed this locally but somehow it did not get committed during the PR. I have a `FlashStorage.cpp` and `FlashStorage.cpp~`, and this is the diff! :+1: 56c56
```c
<   if (CMCC->SR.bit.CSTS) {
---
>   if (CMCC->CR.bit.CSTS) {
```

If you submit a PR I will approve it!

obra, 2021-01-02:
@dhalbert Thanks for the quick confirmation. I've opened PR #47 with the one character change.

PR #47 (2021-01-02): Fix a typo in the SAMD51 cache support added in #38.
Fixes #43
Fixes #46
commit c1bb6d4f (2021-01-02): Fix a typo in the SAMD51 cache support added in #38.

Fixes #43
Fixes #46
closed 2021-01-02 by dhalbert