cmaglie/FlashStorage #49: A lot of errors while compiling FlashStoreAnd Retrieve

Kabron287, 2021-03-22:
Board: Wio Terminal
IDE: Arduino 1.8.13
Compiler output:
`FlashStorage.cpp: In member function void FlashClass::write(const volatile void*, const void*, uint32_t)
 
FlashStorage.cpp: 55:22: error: 'volatile struct NVMCTRL_CTRLB_Type::<unnamed>' has no member named 'MANW
   NVMCTRL->CTRLB.bit.MANW = 1
   ^~~~
 
FlashStorage.cpp: 60:26: error: 'NVMCTRL_CTRLA_CMDEX_KEY' was not declared in this scope
   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_PBC
   ^~~~~~~~~~~~~~~~~~~~~~~
FlashStorage.cpp:60: note  suggested alternative  NVMCTRL_CTRLB_CMDEX_KEY
   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_PBC
   ^~~~~~~~~~~~~~~~~~~~~~~
   NVMCTRL_CTRLB_CMDEX_KEY
 
FlashStorage.cpp: 60:52: error: 'NVMCTRL_CTRLA_CMD_PBC' was not declared in this scope
   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_PBC
   ^~~~~~~~~~~~~~~~~~~~~
FlashStorage.cpp:60: note  suggested alternative  NVMCTRL_CTRLB_CMD_PBC
   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_PBC
   ^~~~~~~~~~~~~~~~~~~~~
   NVMCTRL_CTRLB_CMD_PBC
 
FlashStorage.cpp: 61:33: error: 'volatile struct NVMCTRL_INTFLAG_Type::<unnamed>' has no member named 'READY
   while (NVMCTRL->INTFLAG.bit.READY == 0) { }
   ^~~~~
 
FlashStorage.cpp: 73:52: error: 'NVMCTRL_CTRLA_CMD_WP' was not declared in this scope
   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_WP
   ^~~~~~~~~~~~~~~~~~~~
FlashStorage.cpp:73: note  suggested alternative  NVMCTRL_CTRLB_CMD_WP
   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_WP
   ^~~~~~~~~~~~~~~~~~~~
   NVMCTRL_CTRLB_CMD_WP
 
FlashStorage.cpp: 74:33: error: 'volatile struct NVMCTRL_INTFLAG_Type::<unnamed>' has no member named 'READY
   while (NVMCTRL->INTFLAG.bit.READY == 0) { }
   ^~~~~
FlashStorage.cpp: In member function void FlashClass::erase(const volatile void*)
 
FlashStorage.cpp: 92:24: error: 'NVMCTRL_CTRLA_CMDEX_KEY' was not declared in this scope
   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_ER
   ^~~~~~~~~~~~~~~~~~~~~~~
FlashStorage.cpp:92: note  suggested alternative  NVMCTRL_CTRLB_CMDEX_KEY
   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_ER
   ^~~~~~~~~~~~~~~~~~~~~~~
   NVMCTRL_CTRLB_CMDEX_KEY
 
FlashStorage.cpp: 92:50: error: 'NVMCTRL_CTRLA_CMD_ER' was not declared in this scope
   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_ER
   ^~~~~~~~~~~~~~~~~~~~
FlashStorage.cpp:92: note  suggested alternative  NVMCTRL_CTRLB_CMD_EP
   NVMCTRL->CTRLA.reg = NVMCTRL_CTRLA_CMDEX_KEY | NVMCTRL_CTRLA_CMD_ER
   ^~~~~~~~~~~~~~~~~~~~
   NVMCTRL_CTRLB_CMD_EP
 
Error compiling libraries
FlashStorage.cpp: 93:32: error: 'volatile struct NVMCTRL_INTFLAG_Type::<unnamed>' has no member named 'READY
   while (!NVMCTRL->INTFLAG.bit.READY) { }
   ^~~~~`

LazaroFilm, 2021-06-01:
I get the same result

LazaroFilm, 2021-06-01:
It looks like this library is only compatible with SAMD ARM Cortex M0 boards and throws that error with other boards.